stm32-hotspot/STM32H7-LwIP-Examples #1: Example without RTOS

PAk-CatchFire, 2022-10-08:
Hello.
Is there any project example available for STM32H7 for lwip without RTOS?

Thank you

ABESTM, 2022-10-10:
Hello @PAk-CatchFire,

thank you for your feedback. We will consider to add this example(s) to the repository. It could be quite difficult to create non-FreeRTOS variant for each board, but for one selected it could be possibility.

Please note that in https://github.com/STMicroelectronics/STM32CubeH7 package there are examples without FreeRTOS, although the STM32CubeMX project is not provided.

The main changes (compared to FreeRTOS enabled) should be:
1. The MX_LWIP_Process() needs to be called periodically in the main loop
2. The Ethernet interrupts should be disabled, since the communication is processed in main loop
3. HAL_ETH_Start should be used in ethernetif.c (instead of HAL_ETH_Start_IT)
4. Some APIs of LwIP (e.g. BSD sockets) are not available

Best regards,
Adam Berlinger

PAk-CatchFire, 2022-10-12:
Thank you @ABESTM 
Supposedly, in this post from a ST staff member, this issue was going to be solved in the last release. However, it is not clear to me how to get Lwip working with no FreeRTOS.
Here is a link:
https://community.st.com/s/question/0D53W00001Gi9BoSAJ/ethernet-hal-driver-reworked-by-st-and-available-in-22q1-preview-now-available-on-github?utm_source=pocket_mylist

We are using a STM32H723.
Best Regards

ABESTM, 2022-10-17:
Hello,

as I understood, the mentioned community post is about version 1.10.0 of the https://github.com/STMicroelectronics/STM32CubeH7 package which was already release.
The package contains example for LwIP without FreeRTOS, e.g. here: https://github.com/STMicroelectronics/STM32CubeH7/tree/master/Projects/NUCLEO-H723ZG/Applications/LwIP/LwIP_TCP_Echo_Server
I think this is the best starting point at the moment.

Best regards,
Adam Berlinger

PAk-CatchFire, 2022-10-18:
Thank you @ABESTM 
Is it possible to find this project using a CubeMX  ioc file?
It really speeds up the process.

Thank you

Zzaaacks, 2023-03-24:
Hello there everyone, 
Do you have any updates on the matter?

Thanks for the support