stm32-hotspot/STM32H7-LwIP-Examples #2: STM32H743_Nucleo_ETH example causes a "CORTEX_M7"-Warning in STM32CubeIDE 1.10.1

JumpingCatRabbit, 2022-10-17:
Hello!

**Versions**

 - STM32CubeIDE 1.10.1
 - STM32Cube MCU Package for STM32H7 Series 1.10.0

**Problem**

I tried to compile the STM32H743_Nucleo_ETH example but it causes an warning during code Generation:

[image]

In the IOC-Editor it shows up as green without any issues:

[image]

**How to reproduce**

1. Import the project into eclipse

2. Change the LWIP-IP to force generation of source files

3. Have a look at the popup

**Misc**

The project was _not_ automatically migrated.

Thank you!

ABESTM, 2022-10-17:
Hello @JumpingCatRabbit,

thank you for the feedback. Unfortunately this seems to be a bug in STM32CubeMX. I already reported internally.

It is caused by using default configuration of the MPU during project creation, which is recommended. However configuring additional MPU regions on top of the default one seems to trigger this warning.
Selecting "yes" and ignoring the warning should generated the code and the project should work. Please let me know if you observe some other issues related to this.

Best regards,
Adam Berlinger

JumpingCatRabbit, 2022-10-21:
Can confirm! Implementation works just fine when ignoring the warning.