stm32-hotspot/STM32H7-LwIP-Examples #9: Modifications in LCD_DSI_CmdMode_DoubleBuffer  file->main.c in LCD_DSI_Cmd_DoubleBuffer_CM7

Multifalcon, 2023-12-28:
Line 283

#if (USE_LCD_CTRL_NT35510 > 0)			// ADDED
  NT35510_IO_t			IOCtx;			// ADDED
  static NT35510_Object_t   LCD_Obj;	// ADDED and modified variable name
#endif /* USE_LCD_CTRL_NT35510 */		// ADDED

#if (USE_LCD_CTRL_OTM8009A > 0)			
  OTM8009A_IO_t              IOCtx;
  static OTM8009A_Object_t   LCD_Obj;	// Modified variable name
#endif /* USE_LCD_CTRL_OTM8009A */

Line 383 
IOCtx.Address     = 0;
  IOCtx.GetTick     = BSP_GetTick;
  IOCtx.WriteReg    = DSI_IO_Write;
  IOCtx.ReadReg     = DSI_IO_Read;
  OTM8009A_RegisterBusIO(&LCD_Obj, &IOCtx);				// MOdified variable name
  Lcd_CompObj=(&LCD_Obj);									// MOdified variable name
  OTM8009A_Init(Lcd_CompObj, OTM8009A_COLMOD_RGB888, LCD_ORIENTATION_LANDSCAPE);

ABESTM, 2024-01-02:
Hello @Multifalcon 
I assume this was entered to different repository than intended. None of the Ethernet examples should be using LCD display.
If you agree I will close this issue.
Best regards,
Adam Berlinger

Multifalcon, 2024-01-07:
Absolutely right, this modification, was written to fix a bug found in the LCD display, sorry, you can delete it.
And thanks a lot.
Best regards from Madrid Spain
Manuel Orcera