stm32-hotspot/STM32H7-LwIP-Examples #11: STM32H743_nucleo example issues

efthym, 2024-01-27:
I've tried this example on the above board with code generated from cubeMX 6.10.0 and STM32Cube_FW_H7_V1.11.1 but I used DHCP instead of a fixed IP for the boad. Here's what  I've run into and how I fixed them (all in ethernetif.c)

- In low_level_init(), the semaphore creation code  (lines 299 and 302) is generated with an initial count of 0 from cubeMX. You have it correctly initialized at 1, but this is not marked with an ETH_CODE comment, so it is easy to miss. Unfortunatelly it is outside USER CODE BEGIN/END markers, so with code re-generation it will be lost again. This is crucial, as **the board fails to connect**, at least with DHCP.
- In sys_check_core_locking(), line 983, the breakpoint is run even when there is no lock. I've just added a  `&& lwip_core_lock_holder_thread_id` in the if statement to avoid it. This is minor. Only affects debug mode.
- Two unused variables are declared in lines 107, 108. TxPkt, RxPkt. They are not referenced anywhere. I assume they are leftovers from previous code. This is harmless, I'm mentioning so it doen't confuse someone looking into the code.
- In ethernet_link_thread(), with the above versions of cubeMX and library, the code generates HAL_ETH_Start_IT at line 861, so there is no need for the workaround in lines 806-811 (define of HAL_ETH_Start as HAL_ETH_Start_IT).

ArielElb, 2025-04-03:
> I've tried this example on the above board with code generated from cubeMX 6.10.0 and STM32Cube_FW_H7_V1.11.1 but I used DHCP instead of a fixed IP for the boad. Here's what  I've run into and how I [...]

Can you please share a repo code with the
.ioc file also?

Thanks alot

efthym, 2025-04-03:
I'm afraid I don't have it any more. But I started from the code and .ioc files in this repository and I think the above instructions should help, so you can recreate it.

ArielElb, 2025-04-05:
> I'm afraid I don't have it any more. But I started from the code and .ioc files in this repository and I think the above instructions should help, so you can recreate it.

I will try that ty,
What IDE version you used?
Or it does not metter?