stm32-hotspot/STM32H7-LwIP-Examples #14: [STM32H743_Nucleo_ETH] DMACSR asserts AIS (caused by RBU) after receiving 8 UDP packets

lonelycorn, 2025-03-31:
opened the project with STM32CubeIDE 1.18.0 and migrated to that version.
updated the project with the followings

1. add UDP callback to main.c:
```
static void udp_rx_cb(
    void *arg,
    struct udp_pcb *pcb,
    struct pbuf *p,
    const ip_addr_t *addr,
    u16_t port)
{
	int *c = (int *)arg;
	if (p->len == sizeof(*c))
	{
		*c = *(int *)p->payload;
	}
}
```
2. remove tcp-related code in main.c, and comment-out "locks" in lwipopts.h
3. start UDP, register rx callback, and echo received counter over UDP in `StartDefaultTask()`
```
  /* USER CODE BEGIN 5 */
  // set up UDP
  struct udp_pcb *pcb = udp_new();
  if (NULL == pcb)
  {
      Error_Handler();
  }
  if (ERR_OK != udp_bind(pcb, IP_ADDR_ANY, 5000 /* port */))
  {
      Error_Handler();
  }
  ip4_addr_t remote_addr;
  IP4_ADDR(&remote_addr, 192, 168, 1, 100);
  if (ERR_OK != udp_connect(pcb, &remote_addr, 8080 /* port */))
  {
	  Error_Handler();
  }
  int udp_counter = 0;
  udp_recv(pcb, udp_rx_cb, &udp_counter);
  /* Infinite loop */
  for(;;)
  {
    osDelay(1000);

    /* send UDP packet */
    struct pbuf *p = pbuf_alloc(PBUF_TRANSPORT, sizeof(udp_counter), PBUF_RAM);
    if (NULL == p)
    {
    	Error_Handler();
    }
    memcpy(p->payload, &udp_counter, sizeof(udp_counter));
    err_t err = udp_send(pcb, p);
    if (err != ERR_OK)
    {
    	Error_Handler();
    }
    pbuf_free(p);
  }
  /* USER CODE END 5 */
```
4. update & send counter from a python script via UDP, and observe that the echoed value stops updating from the 9-th packet.
```
[RX] '(0,)' from ('192.168.1.10', 5000)
[TX] '1' to ('192.168.1.10', 5000)
[RX] '(1,)' from ('192.168.1.10', 5000)
[RX] '(1,)' from ('192.168.1.10', 5000)
[TX] '2' to ('192.168.1.10', 5000)
[RX] '(2,)' from ('192.168.1.10', 5000)
[TX] '3' to ('192.168.1.10', 5000)
[RX] '(3,)' from ('192.168.1.10', 5000)
[TX] '4' to ('192.168.1.10', 5000)
[RX] '(4,)' from ('192.168.1.10', 5000)
[TX] '5' to ('192.168.1.10', 5000)
[RX] '(5,)' from ('192.168.1.10', 5000)
[TX] '6' to ('192.168.1.10', 5000)
[RX] '(6,)' from ('192.168.1.10', 5000)
[TX] '7' to ('192.168.1.10', 5000)
[RX] '(7,)' from ('192.168.1.10', 5000)
[TX] '8' to ('192.168.1.10', 5000)
[RX] '(8,)' from ('192.168.1.10', 5000)
[TX] '9' to ('192.168.1.10', 5000)
[RX] '(9,)' from ('192.168.1.10', 5000)
[TX] '10' to ('192.168.1.10', 5000)
[RX] '(9,)' from ('192.168.1.10', 5000)
[TX] '11' to ('192.168.1.10', 5000)
[RX] '(9,)' from ('192.168.1.10', 5000)
[RX] '(9,)' from ('192.168.1.10', 5000)
[RX] '(9,)' from ('192.168.1.10', 5000)
```

ABESTM, 2025-03-31:
Hello,

there should be `pbuf_free(p);` at the end of the `udp_rx_cb` callback, otherwise the received packets are still hold by the application and at the end, the driver doesn't have any free packet in RX buffer to use.

Best regards,
Adam Berlinger

lonelycorn, 2025-04-01:
Can confirm the fix by just adding `pbuf_free(p)` at the end of `udp_rx_cb()`. this is saving me, thanks so much @ABESTM . 

that said, while it's my fault to not have read the API documentation thoroughly, it is somewhat uncommon design for the callback to release the buffer, especially when LWIP knowns to release the buffer when there's not a callback. it feels almost as if "releasing the buffer" is an action of guilt, and it's much preferred for someone else to do it.

[image]

let me close this issue.